hybridgroup/cylon-gpio #40: problem running  Maxbotix demo

jBachalo, 2015-02-13:
This is my exact code. When I run I can see only the first trace. Is there a method or property of *my.maxbotix * that might help in debugging potential hardware, wiring issues?
    var Cylon = require('cylon');
    Cylon.robot({
       connections: {
          edison: { adaptor: 'intel-iot' }
       },
    
    devices: {
        maxbotix: { driver: 'maxbotix', pin:'1' }
     },    
    work: function (my) {
        every((1).seconds(), function () {
            console.log('0 ASSERT  FIRST TRACE 1 second interval my.maxbotix' + my.maxbotix);
            
            my.maxbotix.range(function (data) {
                console.log("1 ASSERT 2ND TRACE range: " + data);
             });
          });
      }
    }).start();

edgarsilva, 2015-02-13:
@jBachalo I see the issue, what is happening is that there is a disconnect (outdated) docs and examples between how this used to work and how it works now.

Basically the `my.maxbotix.range()` only returns a value but does not receives any params, so what you have should work if you do it like this:

```javascript
var Cylon = require('cylon');
    Cylon.robot({
       connections: {
          edison: { adaptor: 'intel-iot' }
    },

devices: {
    maxbotix: { driver: 'maxbotix', pin:'1' }
 },    
work: function (my) {
    var range = 0;
    every((1).seconds(), function () {
        range = my.maxbotix.range();
        console.log('Maxbotix range: ', range);
    });
  }
}).start();
```

edgarsilva, 2015-02-13:
You can also use a similar approach that returns a new value every time there is a read from the maxbotix by doing something like this:

```javascript
var Cylon = require('cylon');

Cylon.robot({
  connections: {
    edison: { adaptor: 'intel-iot' }
  },

  devices: {
    maxbotix: { driver: 'maxbotix', pin:'1' }
  },    
  work: function (my) {
    my.maxbotix.on('range', function(err, range) {
      console.log('Maxbotix range: ', range);
    });

    my.maxbotix.on('rangeCm', function(err, range) { 
      console.log('Maxbotix range in cm: ', range);
    });
  }
}).start();
```

That is a different approach that provides values as fast as the board can read them from the analog input.

jBachalo, 2015-02-13:
Thats it!! thanks

edgarsilva, 2015-02-13:
Cool!, nice to hear it is working now. We are correcting and updating the docs shortly, you can go ahead and close this issue.

closed 2015-02-13 by jBachalo