hybridgroup/cylon-gpio #53: analogRead returning values out of range

NickSchweitzer, 2016-04-17:
I have attached the output of a basic voltage divider to measure the value of a simple pressure sensor to Analog Pin 2 on an Edison Arduino breakout board. I also hooked up a Grove LCD display. The voltages that I read on my DMM range anywhere from 100mv when full pressure applied to 5v when no pressure applied. However, when I run the following code, the values from analogRead appear to be anywhere from about 300 to 9900. Based on the documentation, I was expecting 0 to 1023. Can someone let me know what I'm doing wrong here?

require('cylon').robot({
    connections: { edison: { adaptor: 'intel-iot' } },
    devices: {
        led: { driver: 'led', pin: 13 },
        screen: { driver: "upm-jhd1313m1", connection: "edison" },
        pressure: { driver: 'analogSensor', pin: 2, connection: "edison" },
    },

    updateScreen: function (my, line1, line2) {
        my.screen.setColor(255, 0, 0);
        my.screen.setCursor(0, 0);
        my.screen.write(line1);
        my.screen.setCursor(1, 0);
        my.screen.write(line2);
    },

    work: function (my) {
        var that = this;
        var pressureValue = 0;

        every((0.1).second(), function () {
            my.led.toggle();
        });

        every((0.1).second(), function () {
            pressureValue = my.pressure.analogRead();
            that.updateScreen(my, 'Pressure: ' + pressureValue, '');
        });
    }
}).start();

deadprogram, 2016-04-17:
Hi, @NickSchweitzer 

According to the Intel Edison docs, it supports up to 4096 (12 bit) resolution for Analog input. To determine the exact voltage level from yoru sensor, divide the reference voltage (either 5 or 3.3v) by the number of steps (4096 for 12 bit, 1024 for 10 bit). Multiply the reading from your port by the proper step value, and that will give you the correct V value.

Hope that helps!

NickSchweitzer, 2016-04-17:
Hi @deadprogram - Thanks for the fast response. Couple of things... 

1. Good to know about the increased resolution
2. I mistyped slightly before. The low value I'm seeing is around 3000, not 300
3. The high value I'm seeing is around 9900, which is still higher than that 4096 you quote.

So based on that, something still seems to be goofy here. Any other ideas?

good to know about the cin

deadprogram, 2016-04-17:
I've only used the analog in with very simple sensors on the Edison. 

I'd look at both your circuit, as well as make sure you are working with the pin voltage you are expecting (3.3v or 5v) by checking the switch on the side of the Grove shield.

We use the MRAA library's implementation of analogRead when running on the Edison, so whatever defaults `Mraa.Aio()` uses, is what we are using: https://github.com/hybridgroup/cylon-intel-iot/blob/master/lib/adaptor.js#L163

NickSchweitzer, 2016-04-18:
I discovered my problem. The issue had nothing to do with the Voltage Divider. Instead, what was happening is that the LCD does not overwrite the remaining line if you don't specify the max length of the LCD string. So I need to space pad my messages to overwrite the text. Since I was using the LCD to tell me what the A/D value was, it was throwing me off.

Thanks again for the quick assistance.

closed 2016-04-18 by NickSchweitzer