hybridgroup/cylon-gpio #56: analogWrite doesn't work

JesusIslam, 2017-01-02:
I'm currently trying to do analogWrite(255) to pin 5 of an Arduino Uno, but it doesn't give me voltage reading at all. (it toggle the LED though)

I've tried it using regular arduino code directly on the board and at 255 it gives me around 4.8v.

This is my code:

```
"use strict";

const config = require(__dirname + "/config.json");
config.work = require(__dirname + "/lib/work");

(require("cylon")).robot(config).start();
```

this is the config part
```
{
    "name": "Testbot",
    "connections": {
        "arduino": { 
            "adaptor": "firmata", 
            "port": "/dev/ttyUSB0" 
        }
    },
    "devices": {
        "onboardLED": { 
            "driver": "led", 
            "pin": 13 
        },
        "writer": {
            "driver": "direct-pin",
            "pin": 5
        }
    }
}
```

and this is the work code
```
dev.onboardLED.toggle();
dev.writer.analogWrite(255);
```

deadprogram, 2017-01-02:
When using Firmata pins A4 and A5 are used by the i2c interface. I suggest using pins A0 thru A3.

Hope that helps!

JesusIslam, 2017-01-02:
I am using digital pin 5 (as it says it has PWM support), not analog A5. Or are you saying they are connected?

deadprogram, 2017-01-02:
OK. I think you should use `pwmWrite()` since it will do the mapping you expect (digital pins).

`analogWrite()` does the same thing as `analogRead()` which takes an analog pin number as a param.

JesusIslam, 2017-01-03:
Hold on, I thought `analogWrite()` accept PWM value (0-255) as param? What does it do exactly?

And I think I will try using `pwmWrite()` instead, thanks for the pointer.

deadprogram, 2017-01-03:
The difference is that `analogWrite()` when you pass "5" as a param, it means pin A5. With `pwmWrite()` the param "5" means pin D5.